MeshInspector/MeshLib #4418: How to fill a hole based on its size

guyuezuntinggithub, 2025-04-10:
hi everyone!
I used an API to find out how many holes are in the mesh, but I would like to first determine the size of the holes and then decide whether to patch them based on their size. I hope to get some help, and I would be very grateful
 
this is the api
auto holenum = meshlibmesh.topology.findNumHoles();

Grantim, 2025-04-10:
Hello!

You can use these functions:
```py
from meshlib import mrmeshpy as mm

mesh = mm.loadMesh("path/to/mesh.stl")
holeIds = mesh.topology.findHoleRepresentiveEdges()
for he in holeIds:
    print( mesh.holePerimiter( he ) )
```

guyuezuntinggithub, 2025-04-10:
> Hello!
> 
> You can use these functions:
> 
> from meshlib import mrmeshpy as mm
> 
> mesh = mm.loadMesh("path/to/mesh.stl")
> holeIds = mesh.topology.findHoleRepresentiveEdges()
> for he in holeIds:
>     if mesh.holePerimiter( he ) < 0.1:
>         mm.fillHole( mesh, he ) # fill with default params, or `fillHoleNicelly( mesh, e )`: note that you can specify parameters

very appreciate for your reply! is there C++ API for holePerimiter.

closed 2025-04-10 by guyuezuntinggithub