MeshInspector/MeshLib #4433: Calling the function MR::Mesh::fromTriangles causes a dump

guyuezuntinggithub, 2025-04-12:
Hello everyone, I am compiling code using VS2019 on a Windows 10 system, with MeshLib version MeshLibDist_v2.2.11.51. I wrote a function to convert from Open3D's triangle to MR::Mesh. However, when calling the function MR::Mesh::fromTriangles at the end, it hangs. The error encountered is:
Exception thrown at 0x00007FFAC3488ECC (msvcp140.dll) in PclCloudViewer.exe: 0xC0000005: Access violation reading location 0x0000000000000008.

[image]

code is bellow
MR::Mesh ImageProcessingUtility::open3dsurface2MR(const std::shared_ptr<geometry::TriangleMesh> source_points_ptrhole) {
	MR::Mesh meshlibmesh;
	MR::VertCoords verts;
	MR::Vector3f vert;
	//auto mvalu=loadRes.value();
	auto vertNum = source_points_ptrhole->vertices_.size();
	verts.resize(vertNum);
	auto faceNum = source_points_ptrhole->triangles_.size();
	MR::Triangulation trangles;
	for (int fid = 0; fid < faceNum; fid++) {
		MR::ThreeVertIds threevrtids;
		threevrtids[0] = static_cast<MR::VertId>(source_points_ptrhole->triangles_[fid][0]);
		threevrtids[1] = static_cast<MR::VertId>(source_points_ptrhole->triangles_[fid][1]);
		threevrtids[2] = static_cast<MR::VertId>(source_points_ptrhole->triangles_[fid][2]);
		auto id1 = threevrtids[0];
		auto id2 = threevrtids[1];
		auto id3 = threevrtids[2];
		trangles.push_back(threevrtids);
		verts[id1].x = source_points_ptrhole->vertices_[static_cast<int>(id1)][0];
		verts[id1].y = source_points_ptrhole->vertices_[static_cast<int>(id1)][1];
		verts[id1].z = source_points_ptrhole->vertices_[static_cast<int>(id1)][2];
		verts[id2].x = source_points_ptrhole->vertices_[static_cast<int>(id2)][0];
		verts[id2].y = source_points_ptrhole->vertices_[static_cast<int>(id2)][1];
		verts[id2].z = source_points_ptrhole->vertices_[static_cast<int>(id2)][2];
		verts[id3].x = source_points_ptrhole->vertices_[static_cast<int>(id3)][0];
		verts[id3].y = source_points_ptrhole->vertices_[static_cast<int>(id3)][1];
		verts[id3].z = source_points_ptrhole->vertices_[static_cast<int>(id3)][2];
	}

	meshlibmesh = MR::Mesh::fromTriangles(verts, trangles);
	return meshlibmesh;
}

closed 2025-04-12 by guyuezuntinggithub

guyuezuntinggithub, 2025-07-16:
the whole code is following
`MR::Mesh ImageProcessingUtility::open3dsurface2MR(const std::shared_ptr<geometry::TriangleMesh> source_points_ptrhole) {
	MR::Mesh meshlibmesh;
	MR::VertCoords verts;
	MR::Vector3f vert;
	//auto mvalu=loadRes.value();
	auto vertNum = source_points_ptrhole->vertices_.size();
	verts.resize(vertNum);
	auto faceNum = source_points_ptrhole->triangles_.size();
	MR::Triangulation trangles;
	for (int fid = 0; fid < faceNum; fid++) {
		MR::ThreeVertIds threevrtids;
		threevrtids[0] = static_cast<MR::VertId>(source_points_ptrhole->triangles_[fid][0]);
		threevrtids[1] = static_cast<MR::VertId>(source_points_ptrhole->triangles_[fid][1]);
		threevrtids[2] = static_cast<MR::VertId>(source_points_ptrhole->triangles_[fid][2]);
		auto id1 = threevrtids[0];
		auto id2 = threevrtids[1];
		auto id3 = threevrtids[2];
		trangles.push_back(threevrtids);
		verts[id1].x = source_points_ptrhole->vertices_[static_cast<int>(id1)][0];
		verts[id1].y = source_points_ptrhole->vertices_[static_cast<int>(id1)][1];
		verts[id1].z = source_points_ptrhole->vertices_[static_cast<int>(id1)][2];
		verts[id2].x = source_points_ptrhole->vertices_[static_cast<int>(id2)][0];
		verts[id2].y = source_points_ptrhole->vertices_[static_cast<int>(id2)][1];
		verts[id2].z = source_points_ptrhole->vertices_[static_cast<int>(id2)][2];
		verts[id3].x = source_points_ptrhole->vertices_[static_cast<int>(id3)][0];
		verts[id3].y = source_points_ptrhole->vertices_[static_cast<int>(id3)][1];
		verts[id3].z = source_points_ptrhole->vertices_[static_cast<int>(id3)][2];
	}

	meshlibmesh = MR::Mesh::fromTriangles(verts, trangles);
	return meshlibmesh;
}`

Grantim, 2025-07-16:
Hello!
Could you please provide more details, about error you are facing?

guyuezuntinggithub, 2025-07-16:
ok。the error is
0x0000000000000000 处(位于 PclCloudViewer.exe 中)引发的异常: 0xC0000005: 执行位置 0x0000000000000000 时发生访问冲突。
the .ply is 

[meshwithglb.zip](https://github.com/user-attachments/files/21254487/meshwithglb.zip)

guyuezuntinggithub, 2025-07-16:
I read this file by open3d, and I try to use the function above to change  geometry::TriangleMesh to MR::Mesh ,it core dump at meshlibmesh = MR::Mesh::fromTriangles(verts, trangles);

guyuezuntinggithub, 2025-07-17:
it ocurs new build error "can't open file “spdlog/spdlog.h”"
when I add log code "MR::setupLoggerByDefault();"