MeshInspector/MeshLib #4438: API for calculating mesh normals

guyuezuntinggithub, 2025-04-14:
hi guys！
Is there an API for calculating mesh normals? I want to obtain the normal vector for each triangular face in the mesh.
any reply will be appreciate！

Fedr, 2025-04-14:
Hello,

Yes, we have
```
/// returns a vector with face-normal in every element for valid mesh faces
[[nodiscard]] MRMESH_API FaceNormals computePerFaceNormals( const Mesh & mesh );
```

See `MRMesh/MRMeshNormals.h`

guyuezuntinggithub, 2025-04-15:
> Hello,
> 
> Yes, we have
> 
> ```
> /// returns a vector with face-normal in every element for valid mesh faces
> [[nodiscard]] MRMESH_API FaceNormals computePerFaceNormals( const Mesh & mesh );
> ```
> 
> See `MRMesh/MRMeshNormals.h`

thanks！

closed 2025-04-15 by guyuezuntinggithub