MeshInspector/MeshLib #4501: compensateRadius: support tool direction changes within a certain angle range?

sysuyl, 2025-04-24:
Hello,

The compensateRadius function is very helpful when milling. I found that the current method only supports fixed tool direction. Will it be expanded to support tools with certain orientation changes (for example, tools that can swing a certain angle along the main direction)?

Thanks!

sysuyl, 2025-04-25:
The input data is similar to a crown for restoration. When milling, a tool radius compensation is required for the inner cavity of the crown. It seems that the tool of some five-axis machining machines can swing within a certain angle range in the main direction, such as the z-axis. So I am wondering whether the function here can support this situation? Or have I misunderstood something about this process? In addition, it may not be necessary for the user to specify a direction for each vertex of the object.

Grantim, 2025-04-25:
I see, so function should consider not only one direction for each point, but all possible directions, and find best one (based on amount of extra milling). I think in general it is possible to support, but it requires noticeable effort. 
Also, do  you expect the function to return best direction for each point or are you going to find them with other algorithm?

sysuyl, 2025-04-25:
Hello,

I am not looking for the best direction for each point. I just think that if a point can be cut from another direction (if there is such a device support), the mesh will be less deformed during tool compensation. Such a result may better maintain the mesh shape originally set by the user.
In addition, I think the best direction for a single point may be a bit difficult to define. Maybe it needs to be considered when considering the directionality of all points in the entire mesh or when planning the tool path? But these problems may be more complicated. If my goal is just to generate a mesh that takes tool compensation into account, these results may not be needed now.
In addition, I am just starting to get involved in these. It would be better if you have a deeper understanding of this problem and a more general solution.

Thanks.

Grantim, 2025-04-25:
Thanks for the explanation, we will discuss it with the team and put this task in our roadmap

sysuyl, 2025-04-27:
That's great, I look forward to your progress.