MeshInspector/MeshLib #4748: Using emscripten to package vtk and meshlib into one wasm with undefined symbol: __resumeException

leye123, 2025-06-24:
After compiling vtk locally on the host, use the emscripten version of meshlib and package it using emscripten to produce:

[image]

After trying, the two can compile independently, but in the presence of both, the above error occurs. Is it because VTK and Meshlib are linked to a certain file together? Can you help me answer this?

[CMakeLists.txt](https://github.com/user-attachments/files/20875449/CMakeLists.txt)

oitel, 2025-06-24:
Hello, unfortunately due to the external dependencies' requirements we have to build MeshLib with the exception support, although they are actually not used in the runtime.

leye123, 2025-06-25:
Okay, thank you for your help. I will now organize the project by separating vtk and meshlib separately.