MeshInspector/MeshLib #4874: An outlier point was generated when using `cutMeshByContour()`.

alpinebuster, 2025-07-22:
I'm not sure if this is a bug or I'm using it incorrectly. The curve I use is closed and non-self-intersecting. Here is a specific example curve:

```txt
[
    12.374342918395996,
    -8.568268775939941,
    4.713981628417969,
    12.356268882751465,
    -9.726734161376953,
    4.778738498687744,
    11.647807121276855,
    -11.520567893981934,
    4.8898844718933105,
    10.640680313110352,
    -13.658771514892578,
    5.503952980041504,
    8.986296653747559,
    -14.89557933807373,
    4.826672077178955,
    7.0328264236450195,
    -17.72704315185547,
    5.100575923919678,
    5.3157148361206055,
    -18.36249542236328,
    5.662057876586914,
    2.562943696975708,
    -18.567134857177734,
    5.638067245483398,
    -1.127899169921875,
    -18.045665740966797,
    4.976574420928955,
    -4.701169013977051,
    -16.834407806396484,
    5.414785385131836,
    -8.431866645812988,
    -15.273381233215332,
    4.947081565856934,
    -11.342529296875,
    -13.267176628112793,
    4.123119831085205,
    -11.906951904296875,
    -12.8923921585083,
    4.516324520111084,
    -10.328729629516602,
    -11.59301471710205,
    4.1317572593688965,
    -6.037509918212891,
    -11.114469528198242,
    5.60826301574707,
    -0.2249613255262375,
    -10.612106323242188,
    6.917881011962891,
    5.970120906829834,
    -8.145326614379883,
    6.2469563484191895,
    10.712018013000488,
    -8.319670677185059,
    5.378563404083252,
    12.374342918395996,
    -8.568268775939941,
    4.713981628417969
]
```

The code I used is as follows ([val cutMeshByContourImpl( Mesh& mesh, const std::vector<float>& coordinates )](https://github.com/alpinebuster/meshsdk/blob/bb9d5931c7da2cfb75358ba92b81cc3885af4db1/source/MRJavaScript/MRContoursCut.cpp#L207)). It should be relatively a simple one: I call the function, passing in the `mesh` and the closed curve to obtain the result.

```cpp
val cutMeshByContourImpl( Mesh& mesh, const std::vector<float>& coordinates )
{
	std::vector<Vector3f> polyline;

    int coordinatesLength = coordinates.size();
    if ( coordinatesLength % 3 != 0 ) {
		val obj = val::object();
		obj.set( "success", false );
		obj.set( "error", "Coordinates length must be a multiple of 3!" );

		return obj;
    }
	polyline.reserve( coordinatesLength / 3 );

	for ( size_t i = 0; i < coordinatesLength; i += 3 )
	{
		polyline.emplace_back( Vector3f( coordinates[i], coordinates[i + 1], coordinates[i + 2] ) );
	}

	// FIXME: Why this will cause the inner mesh have a outlier point?
	auto expectedFaceBitSet = cutMeshByContour( mesh, polyline );
	if ( expectedFaceBitSet )
	{
		auto onePart = expectedFaceBitSet.value();

		auto [smallerMesh, largerMesh] = MRJS::returnParts( mesh, onePart );
		val smallerMeshData = MRJS::exportMeshMemoryView( smallerMesh );
		val largerMeshData = MRJS::exportMeshMemoryView( largerMesh );
		val meshData = MRJS::exportMeshMemoryView( mesh );

		val obj = val::object();
		obj.set( "success", true );
		obj.set( "mesh", meshData );
		obj.set( "smallerMesh", smallerMeshData );
		obj.set( "largerMesh", largerMeshData );
		obj.set( "meshInstance", mesh );
		obj.set( "smallerMeshInstance", smallerMesh );
		obj.set( "largerMeshInstance", largerMesh );

		return obj;
	} else {
		std::string error = expectedFaceBitSet.error();

		val obj = val::object();
		obj.set( "success", false );
		obj.set( "error", "cutMeshByContour: " + error );
	
		return obj;
	}
}
```

Results:

[image]
[image]

alpinebuster, 2025-07-22:
FYI: The source mesh I am using is [upper.stl](https://github.com/alpinebuster/meshsdk/blob/bb9d5931c7da2cfb75358ba92b81cc3885af4db1/examples/_assets/upper.stl).

alpinebuster, 2025-07-22:
Yes, using the code you just provided (where the `mesh` instance is loaded from the `fromAnyStl` function exported by Emscripten on the JS side), I can only get the correct result the first time I load a new `stl` model. When I continue to load the same model afterward, I get the erroneous result mentioned above. Could it be an issue with the `Mesh` class exported by Emscripten? However, this `Mesh` class works fine when calling other APIs.

alpinebuster, 2025-07-22:
The lifecycle of the `Mesh` instance should not be an issue, as I called the `delete()` function provided by Emscripten after using it.

Grantim, 2025-07-22:
Its hard to say what is going wrong but it sure can be something on js side. Most suspicious place might be communication between js and filesystem.

alpinebuster, 2025-07-22:
I'll try a bit more.

alpinebuster, 2025-07-22:
Bonjour~
🐛  I have pinpointed the root of the problem, which is unrelated to `MeshLib`. This is an issue related to the **memory view** in Emscripten. The function [val exportMeshMemoryView( const Mesh& meshToExport )](https://github.com/alpinebuster/meshsdk/blob/e3eadcf8e580efdf00d481aa1e2483b5ff7cc0a8/source/MRJavaScript/MRUtils.cpp#L285-L286) uses a memory view. The specific reason is still unclear to me, but this leads to an additional vertex in local coordinate system. The corresponding code block is as follows:

```cpp
// === Export point data ===
const auto& points_ = meshToExport.points;
size_t pointCount = points_.size();
size_t vertexElementCount = pointCount * 3;
const float* pointDataPtr = reinterpret_cast<const float*>( points_.data() );

// Use `typed_memory_view()` for vertices
val pointsArray = val( typed_memory_view(
    vertexElementCount, 
    pointDataPtr
) );

// === Export triangle data ===
const auto& tris_ = meshToExport.topology.getTriangulation();
size_t triangleCount = tris_.size();
size_t triElementCount = triangleCount * 3;
// NOTE:
// 
//  uint32_t*   	-> Uint32Array, we need `Uint32Array` for threejs
//  int*	        -> Int32Array
// 
const uint32_t* triDataPtr = reinterpret_cast<const uint32_t*>( tris_.data() );

/// NOTE: V3 - Working & Faster than V1
val triangleArray = val::global( "Uint32Array" ).new_( triElementCount );
val triangleView_ = val( typed_memory_view( triElementCount, triDataPtr ) ); // Use `typed_memory_view()` for triangles
triangleArray.call<void>( "set", triangleView_ );
///
```

PS: These minor issues can occasionally be quite time-consuming, particularly for a C++ newbie. This further underscores the considerable effort invested by the authors of Meshlib. Much respect!👍👍👍

alpinebuster, 2025-07-23:
> Please note that `meshToExport.topology.getTriangulation()` produces temporary object that is freed at the end of `tris_` scope, so `triangleView_ ` is valid only while `tris_` are in scope

Got it, thks!