MeshInspector/MeshLib #4883: Mesh Repair

zhangjiangzhao123, 2025-07-24:
I was pleasantly surprised to find that MeshLib is very suitable for my project, but I have some questions regarding mesh repair. 

**Question 1**
When my model has some holes and bad orientations:

Method 1: I first repair the bad orientations and then repair the holes, but the resulting model is very strange and seems to have more issues.

Method 2: I first repair the bad orientations, then export the model, re-import this model, and then repair the holes. The resulting model is correct with no errors.

I'm wondering if there are other necessary operations missing in Method 1. I tested using MeshInspector (2.5.2.208) and found that simply repairing the flipped normals does not change the model's boundaries, but the expected effect is that they should have merged. Could this be the cause of the problem?

The following is the test program and video.
[test.zip](https://github.com/user-attachments/files/21401024/test.zip)

**Question 2**
It seems that the function `findDisorientedFaces` in `MRMesh/MRMeshFixer.h` does not provide an interface in `MRMeshC/MRMeshFixer.h`, and there are also some C++functions that do not provide a C interface. Is this intentional?

Grantim, 2025-07-24:
Hello!

Thanks for detailed description, here are two issues with your code:
1. Fixing of disorientation that you use is not safe (`flipOrientation` requires full components), you need code like this:
```c++
duplicateMultiHoleVertices( mesh ); // needed for better separation of components for disorientation fix
auto res = findDisorientedFaces( mesh, params );
if ( !res.has_value() )
    return unexpected( std::move( res.error() ) );

// before flipping it is required to expand selection to full component
MeshComponents::ExpandToComponentsParams expParams;
expParams.incidence = MeshComponents::PerVertex;
expParams.coverRatio = expandRatio_; // 0.5 by default
auto expandedRes = MeshComponents::expandToComponents( mesh, *res, expParams );
if ( !expandedRes.has_value() )
    return unexpected( std::move( expandedRes.error() ) );

// flip whole components
auto ued = getIncidentEdges( mesh.topology, *expandedRes );
mesh.topology.flipOrientation( &ued );

// need to invalidate caches after fliping topology
mesh.invalidateCaches( false );
```

2. You miss `MeshBuilder::uniteCloseVertices( mesh, 0.0f, true );` call before filling holes, and **re-importing from stl format does that for you** because stl format does not store actual topology and vertices with identical coordinates are always merged on inport.

---

>Question 2
It seems that the function findDisorientedFaces in MRMesh/MRMeshFixer.h does not provide an interface in MRMeshC/MRMeshFixer.h, and there are also some C++functions that do not provide a C interface. Is this intentional?

Right now C API is not full, it is manually update for most popular functions, but are working on autogeneration that will cover all meshlib functionality.

zhangjiangzhao123, 2025-07-24:
Thank you for your reply. Now it can achieve the correct result.

zhangjiangzhao123, 2025-07-25:
Thank you again. I checked my C++ program once more, and everything is perfect.

zhangjiangzhao123, 2025-07-30:
I have reopened this issue, this time focusing on performance. When testing a model over 1GB, detecting inverted normals consumes too much memory and time. The same happens when using MeshInspector. However, when using other professional software, the performance and memory usage are acceptable. I will send the model to the designated email address and hope it can be tested and that I can receive some suggestions.

closed 2025-07-31 by zhangjiangzhao123