MeshInspector/MeshLib #4917: How can I calculate the minimum distance between two meshes and return the nearest point in C#?

6ther, 2025-08-02:


Fedr, 2025-08-03:
In C++ we have a function for this in `MeshLib/source/MRMesh/MRMeshMeshDistance.h`:
```c++
struct MeshMeshDistanceResult
{
    /// two closest points: from meshes A and B respectively
    PointOnFace a, b;
    /// squared distance between a and b
    float distSq = 0;
};

/**
 * \brief computes minimal distance between two meshes or two mesh regions
 * \param rigidB2A rigid transformation from B-mesh space to A mesh space, nullptr considered as identity transformation
 * \param upDistLimitSq upper limit on the distance in question, if the real distance is larger than the function exists returning upDistLimitSq and no valid points
 */
MRMESH_API MeshMeshDistanceResult findDistance( const MeshPart & a, const MeshPart & b,
    const AffineXf3f * rigidB2A = nullptr, float upDistLimitSq = FLT_MAX );
```

but it is not exposed in C# yet. We will try to add it in the nearest future.

6ther, 2025-08-03:
I find that "FindSignedDistances" function  is exposed in C#, but I do not know  if it is used to calculate minimum distance because it's return result is a float list.

Fedr, 2025-08-03:
Yes, in C# we have
```c#
/// Computes signed distances from valid vertices of test mesh to the closest point on the reference mesh:
/// positive value - outside reference mesh, negative - inside reference mesh;
/// this method can return wrong sign if the closest point is located on self-intersecting part of the mesh
unsafe public static VertScalars FindSignedDistances(Mesh refMesh, Mesh mesh, MeshProjectionParameters parameters)
```
but it computes not one closest distance, but the distance from each vertex of `mesh` to `refMesh`. That is why the return type is float list. You can manually then search for the closest distance in the list. And then project again the corresponding vertex of `mesh` to `refMesh` to find the closest point on the later.

6ther, 2025-08-04:
I can search for the closest distance in the list, but I do not know whitch corresponding vertex of mesh calcualte the closest distance.

Fedr, 2025-08-04:
If you find that the minimum distance is in `i`th element of the list, then the corresponding vertex has ID `i`.

6ther, 2025-08-04:
I see. There is a problem, that is, the minimum distance between two meshes may not necessarily be at the vertex, isn't it?

Fedr, 2025-08-04:
Indeed, in general the minimal distance between two triangular surfaces can be achieved:
1) if one of the closest points is located in a vertex (and the other in any place of the closest triangle on the other mesh),
2) if both of the closest points are located on the edges of both meshes.

The current solution present in C# covers only 1). But it will give you a good approximation from above (the actual closest distance can be only smaller).

The other function that is present in C++ for now, will cover the general case after exposing in C#.

6ther, 2025-08-06:
thanks a lot.

6ther, 2025-08-13:
[image]
I find that new version is released in  nuget and I update, but I do not find Lib folder and related dll.

Grantim, 2025-08-13:
v3.0.7.237 was corrupted, we have just uploaded v3.0.7.238, it should be ok, could you please verify?

6ther, 2025-08-14:
yes, thanks.

6ther, 2025-08-19:
Many dependency libraries are missing：
[image]

[image]
What should I do?

Fedr, 2026-09-23:
Hi, a status update on this issue:

- `Mesh.FindDistance` (the minimum distance between two meshes, plus the closest points) has been available in the C# NuGet package since v3.0.7.238.
- The Linux native libraries in the NuGet package need glibc ≥ 2.28 (manylinux_2_28). That means RHEL / Rocky / AlmaLinux 8 or newer, Ubuntu 20.04+, Debian 10+. CentOS 7 (glibc 2.17, end-of-life since June 2024) can't be supported. Please move the container to a newer base image, e.g. `rockylinux:8`.
- On a minimal/headless image you may also need to install the system OpenGL library (`dnf install mesa-libGL`), which the package doesn't bundle.

I'm closing the issue since the original question is resolved. Feel free to reopen if you hit problems on a supported distribution.

closed 2026-09-23 by Fedr