MeshInspector/MeshLib #5004: 2 or more viewports: setting camera has a frame delay

zag-taishev-adalisk, 2025-08-25:
In my code I have 2 viewports and set up an object and camera while camera rotation and translation so,
that this object on the scene stays remaining at the same place to user on both viewports

To make it I set up object's Xf for the 1st viewport and set up camera on the 2nd viewport and expect that the object is not moving

However when I do it I can see object dragging and shaking when I rotate and translate camera

Example test branch with test only code for Rotator Plugin (test/viewport-sync-repro):
[](https://github.com/MeshInspector/MeshLib/tree/test/viewport-sync-repro) 

This issue can be fixed from this code, that was done by Fedor, but it looks like a workaround (test/viewport-sync): 
[](https://github.com/MeshInspector/MeshLib/tree/test/viewport-sync)

Grantim, 2025-08-25:
https://github.com/MeshInspector/MeshLib/pull/5006
This signal should help with manipulating viewports without one frame latency

Grantim, 2025-08-25:
Please check updated [test/viewport-sync-repro](https://github.com/MeshInspector/MeshLib/tree/test/viewport-sync-repro) branch, it has both `preSetupView` slot, to update viewports without frame delay, and `preDraw_` slot for label xf compensation

closed 2025-08-25 by Grantim